iotempire/iotempower #17: Error in adaption mode

LWlook, 2019-11-15:
[image]

If I want to adapt a node with the powershell i get the error like in the picture.

ulno, 2019-11-16:
Ah, that's pretty interesting. I use terminal-kit (a node.js module, https://github.com/cronvel/terminal-kit) to do the menu stuff in the command line. I assume that power-shell has some weird stuff in its terminal-files going on that prevents this from work.

Can you test if the basic demos of terminal-kit work in powershell?

For continue the use of powershell, you can just use the iot command line programs instead of user menu.

To adopt:
dongle scan
then copy the full name of the node to adopt, then
adopt full-name-of-node

To adopt or initialize via serial:
adopt serial

To deploy, just:
deploy

Does that help?

Best,
Ulrich

ulno, 2024-04-22:
I think this is fixed - can anybody of our windows users check on that @mbz4?