iotempire/iotempower #27: Changing Port used for MQTT in ESP8266

Jakob-B, 2021-11-17:
It appears that the MQTT port (1883) used by ESP8266 is hardcoded. However, in some cases it would be useful to change the port used for MQTT, ideally also using the `node.conf` or `system.conf` file. (I know that 1883 is a standardized port for MQTT, but would be a nice feature that could come in handy i.e. for debugging)

PLCMercenary, 2023-03-18:
is this something I could contribute some effort into? I dont see any "help wanted" or "good first issue" tags on any open issues. Is a community contributor OK for this project and task?

ulno, 2023-03-18:
Thanks for reaching out, PLC-Mercenary.

PRs are very welcome. The port parameter would not be one of the most
pressing issues, but it would be a good start to contribute, indeed.
The framework has not garnered much attention outside of our education
scene, so I wonder how you stumbled upon it - though I have seen
Apart from fixing bugs, we of course also need lots of help with
documentation, organization, maybe some marketing and/or success reports,
and of course even some technology exploration - like mesh network
integration, new sensors/actors, maybe a better UI (even opening some of
these ideas as issues would help) - also having a pressing issue with
connecting esp32s to the pi-accesspoint that we even have not filed as
issue yet as we don't understand what is actually going wrong exactly.

-- ulno

ulno, 2025-12-09:
Thank you for this suggestion. We're restructuring our issue management approach to focus on sustainable and maintainable work. This issue hasn't had activity in more than 12 months.

If you'd like to work on this or if it's become more important to you, please comment here, and we can reopen it with support. Otherwise, feel free to open a fresh issue if this becomes a priority in the future.

See our project restructuring discussion in #192 for context.