iotempire/iotempower #76: new landing page (first page after opening gateway) with new tabs opening when clicking links

ulno, 2024-04-20:


mbz4, 2024-04-22:
Potentially duplicate w/ https://github.com/iotempire/iotempower/issues/89?

mbz4, 2024-04-22:
Oh, this refers to when running `web_starter` and opening `localhost:40080` in the browser, the page that displays - when clicking on links there, we would like those links to open a new tab instead of current tab.

mbz4, 2024-04-22:
This post seems relevant to this issue: https://stackoverflow.com/questions/11716781/open-a-link-in-a-new-window-in-restructuredtext

mbz4, 2024-05-23:
So... we don't have direct control over attributes in Sphinx, like `target="_blank"`...

In order to configure links to open in a new tab in .rst docs we need to adjust Sphinx and use a custom JavaScript solution that automatically adds this attribute to external links when the HTML is loaded in the browser

```reStructuredText
Resources
---------

Product page: `M5Stack Stick C <https://shop.m5stack.com/products/stick-c?variant=43982750843137>`_

```

```javascript
document.addEventListener('DOMContentLoaded', function() {
    var links = document.querySelectorAll('a[href]');
    links.forEach(function(link) {
        if (link.hostname !== window.location.hostname) {
            link.target = '_blank';
        }
    });
});
```
But this means that all links will open in a new tab.
@ulno would this be desirable behaviour?

mbz4, 2024-05-23:
Testing this solution:

```python
extensions = [
    "sphinx_new_tab_link",
]
```
But it wants this:

```python
pip install sphinx-new-tab-link
```

More here: [Guide of sphinx-new-tab-link](https://ftnext.github.io/sphinx-new-tab-link/guide.en.html)

Learned about it from here: [Stack Overflow: Add open-in-new-tab links in Sphinx/reStructuredText](https://stackoverflow.com/questions/25583581/add-open-in-new-tab-links-in-sphinx-restructuredtext)

mbz4, 2024-05-23:
This worked out of the box. Navbar links behave the same.
@ulno not sure how to add this pip install to run automagically when installing iotempower...

mbz4, 2024-05-23:
It doesn't feel that nice to have new tabs continually open...

mbz4, 2024-05-23:
Having severe caching issues, suspect hidden symlinks to be the culprit:

[image]

[image]

[image]

[image]

In other news...
This line makes it possible to include custom `.js` and `.css` files and further customise the look and feel of the pages:
```python
#html_static_path = ['_static']
```
@Natomac

mbz4, 2024-05-23:
Resolved previous issue with hard cache clear.

New problem: when viewing on mobile, the little burger button does not work - no drop down or action... this means users have to switch to a desktop site to access navbar items.

[image]

mbz4, 2024-05-23:
Looking into this too:
[image]

ulno, 2024-05-24:
Can't we add a post parser to add the target blank and use a convention ton
achieve the n parser to detect it? Seems like we could wrap the actual
parser. But also not opposed tom potentially switch to nikola or gohugo for
website generation.

Ulrich

mbz4, 2024-05-27:
or we can try another site generator - [VuePress](https://vuepress.vuejs.org/) maybe?