iotempire/iotempower #81: brand protection actions

ulno, 2024-04-20:


mbz4, 2024-04-22:
Yes, getting in touch with the legal department would help. One safety net is to make sure they understand this is an open-source framework.

From there on, start with defining the IoTempower framework.

Official git readme:
```txt
IoTempower is a framework and environment for empowering everyone to explore and develop for the Internet of Things (IoT) -- tinkerers, makers, programmers, hobbyists, students, artists, and professionals alike. It has a special focus on education and is intended to support classes to teach Internet of Things (IoT) and home automation.

However, it also supports existing IoT deployments and brings mechanisms for over the air (OTA) updates and automatic multi-device deployment. All with a permissible license (and using only libraries with permissable licenses).
```

Next, need to clarify what is licensed how in this framework: 
- **documentation**: we have our own docs, unclear what license we use for those
  - graphics, marketing material etc license? 
- **hardware**: we use off-the-shelf 3rd party hardware licenses
- **software**: MIT permissive

Then need to gauge market presence. 
Here it would help to have proof of commercial ventures that employ the IoTempower framework in any way and/or presently or in the past infringed on any naming convention used by the framework. Since the MIT license requires a reference to the original source(s) but free commercial use, this should be straightforward to find.

Suppose there is a clear need to protect the intangible assets side of the framework, for example things other than software, hardware or documentation: specific designs, patterns, structure of repos, code semantics; anything that can be directly associated with IoTempower could be protected in some way to ensure easier and more level recognition.

For trademarking though, there's one recommended option available: [EUIPO EU Trademarking](https://www.euipo.europa.eu/en/discover-ip/sme-fund?gad_source=1) as this can be reimbursed 75 - 90% and automagically applies in each EU country and in cases extends to other countries as trade partners, perhaps even US.

In case IoTempower needs this level of brand protection. I would consult an expert - there's a few in Tartu, including the former minister of social affairs who contributed to EU legislation directly: [Marju Lauristin](https://en.wikipedia.org/wiki/Marju_Lauristin).
She was part of European Parliament where they worked on the e-Privacy regulation.

She often invites persons to come to her with IPR related questions and is a frequent guest speaker at the UT Research Integrity course.

Before reaching out, it may be helpful to provide 2-3 examples of similar frameworks and how they implemented brand protections:
- [ESPhome.io](https://esphome.io/) ==> Nabu Casa (author) ==> https://www.nabucasa.com/tos/  | it has a logo and docs and software and terms of service so they likely took steps to implement something to protect these assets + its the closest most obvious competitor framework
- [Home Assistant](https://www.home-assistant.io/) also has lots of interesting graphical material, which certainly must have some protections in place given the platforms commercial use-cases and online presence + how they market themselves... see the socials on github for example: https://github.com/home-assistant - these are all intangible assets and are very valuable as well
- other open-source frameworks, platforms, products, especially ones with the MIT license are great case study opportunities
  - [Godot Engine](https://godotengine.org/)
  - [Blender](https://www.blender.org/) (though GPL licensed) but note that [Blender Marketplace](https://blendermarket.com/) is a separate entity which directly supports the Blender product, brand etc 

There's for sure many more examples but my take away from all this is simple:
- what's the impact of IoTempower and its unique use-cases, features; value
- how is it marketed, presented (and how not)
- how has it been used commercially
- Finally, with this picture painted, try to protect these assets:
  - helps to protect the image, brand of the platform (think IoTknit example - had to rename a feature!)
  - helps clarify what's the scope of free to use
    - we put the time down to make it work
    - in order to communicate this publicly, need something distinct to make that possible  
 
There's a great US case law derived explanation for how and why to implement some protections, guaranteeing the product/service is as expected even when there's no guarantee (MIT license) that it works: [Open Source Casebook - Trademarks in Open Source](https://google.github.io/opencasebook/trademarks/). 
I think there's a lot of exciting prospects that can emerge from pursuing protecting IoTempower's assets: more collabs, more professional dev cycles, recognition, awareness, partnerships etc

Farnaz03, 2024-04-22:
Some metrics you can use to show the value or 'effectiveness' of this open-source project (as compared to others) are:
[image]
src: "Leading Effective Engineering Teams" by Addy Osmani, a senior staff engineering manager working on Google Chrome with over 10 years of experience at Google. (https://learning.oreilly.com/library/view/leading-effective-engineering/9781098148232/ch03.html)

mbz4, 2024-09-04:
@ulno IoTempower/IoTempire trademark registered in EE?

ulno, 2024-09-04:
yes, but no resources (mainly time) to continue on that process
TODO: non profit founding, EU-wide protection