iotempire/iotempower #86: assign all remaining iotempower issues here into the project

ulno, 2024-04-20:


ulno, 2024-04-21:
estimate [ulno]: 1h

This shoudl be doen soon to not disconnect things

mbz4, 2024-04-22:
Added workflow automation such that if a PR, or open issue are added, they show up in the Project. If an existing issue is open but not in the project, it won't appear unless the project gets an update (ie, add label, manually add to project).

[image]

Enabled this workflow.
For new issues, they head to backlog.
For urgent issues have separate item in project already: https://github.com/iotempire/iotempower/issues/97
In short, this one requires custom automation and can be achieved using a new workflow file added to `.github/workflows` directory or repo root directory or others... more in that item. But the idea with urgent is that it should be moved directly to `In Progress` status, to receive priority processing/review/attention/...

These are some thoughts.

mbz4, 2024-05-22:
@ulno

closed 2024-05-22 by mbz4