iotempire/iotempower #89: New iot menu or even default shell as user menu evtl. in python

ulno, 2024-04-21:
The iot menu is quite popular, so we should think if we could do everything command line based with it (maybe offer simple shell escape right on front page).
I also don't like that it is written in java script. There exists a prototype in python using prompt_toolkit in bin - after adding prompt_toolkit with ``pip install prompt_toolkit`` (inside of the it environment), you can test it with:
```
cd "$HOME/iot/bin"
python user_menu-python.py
```

However it was once very slow on the pi, should be tested on pi and pi zero.

ulno, 2024-04-21:
estimate [ulno]: 160h

mbz4, 2024-04-22:
https://github.com/Textualize/textual looks promising!
[image]
Feels almost like a full docs page running in the CLI - in fact Textual references [Frogmouth](https://github.com/textualize/frogmouth) for visualising markdown from terminal in addition with directly mentioning SBC friendliness.
So, it might replace/mirror web_starter?
Two things to ensure are in sync + man pages using the same .rst docs.
Guess it might be something like: punch in `iot` then this _textual/frogmouth_ page pops up in terminal, simultaneously starting `mqtt` & `web_starter`, but also showing both indicated started (like a dashboard)?

stenoe, 2024-04-22:
There are also some options in Go available like  [tview](https://github.com/rivo/tview) or something like [BubbleTea](https://github.com/charmbracelet/bubbletea). Go can be an asset if there is need for fast and lean executables, it can crosscompile for numerous platforms including Pi's and Pi zeros etc. and no need for python libs install.