iotempire/iotempower #93: Glossary menu landing Page with IoTempower commands

ulno, 2024-04-21:
Let's create a glossary with all the commands that can be used in the IoTempower environment. 
mqtt_listen etc
Tool support

ulno, 2024-04-21:
Use https://github.com/iotempire/iotempower/blob/master/doc/tool-support.rst and help output of commands as reference - and potentially unify.

mbz4, 2024-04-22:
Potentially duplicate issue w/ https://github.com/iotempire/iotempower/issues/89?

Need clarification:
Is this is a feature request to include a new command in `iot` environment which can **printout** the contents of `tool-support.rst`?

mbz4, 2024-05-22:
@ulno is this what you had in mind?

[image]