iotempire/iotempower #98: SCD4X Driver & Docs

mbz4, 2024-04-23:


mbz4, 2024-04-23:
Update: ended up testing daisy chained with SGP30, SHT30, ENS160 and AHT2X sensors together on the same board and same platformio project, felt convenient as the SCD4x official i2c library has a few calibration and offset features so it made sense to merge the project 

[image]

mbz4, 2024-04-24:
Started with this library: [git: DFRobot_SCD4X](https://github.com/DFRobot/DFRobot_SCD4X)

mbz4, 2024-04-25:
Tried available libraries for async measurement support but the sensor is not robust (plug/unplug test - measurements differ too much, unreliable). Digging deeper into sensirion scd4x library...

mbz4, 2024-05-21:
There be whales here! 🥇 

[image]

ulno, 2024-05-21:
Maybe link the bias ribbit finding here?
Maybe it could be solved with an IoTempower filter?
What about an initial PR to encourage more testers?

mbz4, 2024-05-21:
> Maybe link the bias ribbit finding here?

[GMAC_2024_-_Ribbit_Network_Poster_-_White_BG_48_x_36_in.pdf](https://github.com/iotempire/iotempower/files/15393718/GMAC_2024_-_Ribbit_Network_Poster_-_White_BG_48_x_36_in.pdf)

+ Ribbit issue detailing their decision process narrowing down sensor options: https://github.com/Ribbit-Network/ribbit-network-frog-hardware/issues/103

> Maybe it could be solved with an IoTempower filter?

ribbit in the poster don't really state it as a linear steady state bias error... so do we just need to the `add the difference to each measurement to offset the error from reference sensor` (vaisala)? ... i think we need to test this over a variety of temp, humidity ranges + CO2 concentration levels to really show the behaviour; but the datasheet does show deviations in specific ranges for the scd4x: https://sensirion.com/media/documents/48C4B7FB/64C134E7/Sensirion_SCD4x_Datasheet.pdf

> What about an initial PR to encourage more testers?

Can do, but will have to then regress the updated landing page architecture diagram.

mbz4, 2024-05-21:
https://github.com/iotempire/iotempower/pull/118

closed 2024-05-22 by mbz4