iotempire/iotempower #106: IoT cart

ulno, 2024-05-02:
This task is stuck as I need to find an Estonian speaker to go with me to our local Delta furniture guy

ulno, 2024-05-02:
Imho, the only one in our team who speaks a  bit Estonian is @stenoe - but he does not work in Delta.

ulno, 2024-05-02:
Eero is going to help

mbz4, 2024-05-02:
#### can do some fancy labels when done :)

##### Icon:
[image]

##### GPT:

[image]

ulno, 2024-05-03:
wheels are greenlit 2h of project are already used up, so 8h with orderign and moutning is probably appropriate - just for a future reminder: such a task consumes at least 8 hours work time

ulno, 2024-05-03:
ordered handle and wheels, should be here by end of may

mbz4, 2024-05-21:
ToDo:
- [ ] Cupholder
- [ ] Logo Spray
- [ ] Built-In network switch
- [ ] Built-In power extension cord
- [ ] shock dampers (can try window sealant)

mbz4, 2024-05-21:
Otherwise this task done for now, can prepare second task for remaining To-Do

closed 2024-05-22 by mbz4