iotempire/iotempower #115: Design Wifi configuration menu

Natomac, 2024-05-14:
In IoT Menu
- In a configuration tab (exists already)
- Have a project wifi configuration. (we already have it. The one that updates the system.conf file in iot-systems
- Have a config gateway WiFi tab for more global settings:
- Prompts the user to create their SSID and Password. (!!! NEED TO THINK ABOUT A SECURITY SOLUTION)
- Populates a wifi.txt master config file (like in the PI). A general network settings file.  It also auto updates the IoTempower.config file. !!! right now it creates The iotempower.config.test and populates it with the inputs from user.
- Prompts the question: Do you want to use the internal wifi chip or an external router (with OpenWRT) as gateway? Provide suggestions and instructions. ex. Not all internal chips work. External router is advisable, etc. 
- If they choose internal:
  - Run a script that checks if their machine has a usable wifi chip. 
    - If yes: Then update Network Manager (or hostapd) with the credentials from the wifi.txt
    - If no: Show documentation that educates the user about which cards can work. Advise them to use an external router.
- If they choose external router:
  - Prompt the user - Do you want to configure it now? Which one do you want to do it manually or use our image?
    - If the choose the image option: We auto create a OpenWRT image with Mosquito and a custom settings file. 
      The idea is that we generate a full image that we can flash in the router. I comes with everything. Config, services. (ANTON)
   -  If they choose the manual option: We provide same simple instructions like. Install mosquito nossl, create your hotspot.
- Finally, after everything is set up, run a script that checks if everything is configured okay.
- If not trouble shoot??

Ideias for the external router option? 
- Configuration file for mango or  any openwrt /UCI
- Sritp file with SSH with UCI commands one by one
- wait for it to boot again and make sure computer can talk to the mango.

Natomac, 2024-05-14:
Have been working on easy menu set up with Anton for the external openwrt router. It is working well on the Mango router

tonysln, 2024-05-14:
Will test on the DIY HLK-7628N board too.

tonysln, 2024-05-15:
Also, possibly useful to switch from Dropbear to OpenSSH (with SFTP): [Guide](https://openwrt.org/docs/guide-user/services/ssh/openssh_instead_dropbear), might resolve some of the issues we had when attempting to connect for the first time.

ulno, 2025-12-09:
it's mainly done and accessible from iot menu - no activity for >12months