iotempire/iotempower #120: add milestones

mbz4, 2024-05-22:
# You can use milestones to track progress on groups of issues or pull requests in a repository.

https://docs.github.com/en/issues/using-labels-and-milestones-to-track-work/about-milestones?apiVersion=2022-11-28

How to create a milestone?

https://docs.github.com/en/issues/using-labels-and-milestones-to-track-work/creating-and-editing-milestones-for-issues-and-pull-requests

Next to the search field, click **Milestones**.

[image]

Then, choose one of these options:

- To create a new milestone, click **New Milestone**.

- To edit a milestone, next to the milestone you want to edit, click **Edit**.
 
[image]

Then, type the _milestone's title, description, or other changes_, and click **Create milestone or Save changes**. Milestones will render Markdown syntax.

mbz4, 2024-05-22:
A few suggested milestones:

1. Support CO2 sensors
- [ ] SCD
- [ ] ENS
- [ ] SGP
- [ ] MH-Z19
- [ ] Vaisala

2. protocol bridging
- [ ] UART
- [ ] LoRa
- [ ] BLE
- [ ] SPI?
- [ ] BATMAN!

3. WiFi Serial Flashing Daemon
- [ ] web serial (init flash node(s) from browser)
- [ ] remote serial (init flash node(s) from anywhere) 

4. InfluxDB integration
- [ ] database for storing data
- [ ] maybe also grafana/frontend?
- [ ] SD card support
- [ ] key value storage

5. graphical overhaul
- [ ] landing page
- [ ] new architecture diagram
- [ ] map out all features (class diagram)
- [ ] logo, branding

6. Git repo automation
- [ ] add testing workflow
- [ ] add build badge (passing/failing etc)
- [ ] add sponsor feature

ulno, 2025-12-09:
Thank you for this suggestion. We're restructuring our issue management approach to focus on sustainable and maintainable work. This issue hasn't had activity in more than 12 months.

If you'd like to work on this or if it's become more important to you, please comment here, and we can reopen it with support. Otherwise, feel free to open a fresh issue if this becomes a priority in the future.

See our project restructuring discussion in #192 for context.