iotempire/iotempower #135: MH-Z19 (PWM) CO2 sensor support package

mbz4, 2024-07-04:


mbz4, 2024-07-04:
PWM Async example working:
[image]

On to preparing iotempower driver draft...

ulno, 2024-07-04:
What prevents you to use the built in pwm functionality of IoTempower?

mbz4, 2024-07-04:
> What prevents you from using the built-in pwm functionality of IoTempower? This would then allow to run on all esp32 and esp8266 platforms?

docs show this PWM output definition for ppm measurements:
[image]

will test w/ native iotemp pwm library first

ulno, 2024-07-04:
ah it is for reading pwm - not writing
Reading might be impossible until the buffered async reader works - so you
might need to help me fix that one first.
But all depends on timing.

So the serial communication port is just bogus and useless?

> Message ID: ***@***.***>
>

mbz4, 2024-07-04:
@ulno `pwm(CO2, D2, 2000);` gave this:

[image]

> ah it is for reading pwm - not writing Reading might be impossible until the buffered async reader works - so you might need to help me fix that one first. But all depends on timing. So the serial communication port is just bogus and useless?

ok

i figured i'd try w/ PWM since serial also unsupported
guess one thing at a time, will push back to Todo/Backlog